MozOpenHard/CHIRIMEN #2: CHIRIMEN用B2Gリポジトリを用意する

naokisekiguchi, 2015-04-20:
https://github.com/flingone/B2G-FlingOne.gitをベースに用意する。
対象のFirefoxOSのバージョンは2.1とする。

YusukeYamamoto, 2015-04-20:
メモ

# 概要
以下のユースケースに対応出来るようにしたい。

1. $ git clone https://github.com/MozOpenHard/CHIRIMEN/xxxxx/B2G.git
2. $ cd B2G
3. $ ./config.sh chirimen
4. 〜しばし待つ〜
5. $ ./build.sh
6. 〜専用の焼き焼きツールで焼く〜

# 対応が必要な項目案
今思いつく範囲で・・・
* [x] https://github.com/MozOpenHard/CHIRIMEN/xxxxx/B2G.git の作成  
https://github.com/mozilla-b2g/B2G から Fork してくる  
→ https://github.com/MozOpenHard/B2G これで良いぽい?
* [ ] config.sh に chirimen のデバイス名の追加
→https://github.com/mozilla-b2g/B2G/blob/master/config.sh に追加する
* [ ] chirimen用の xml の追加  と https://github.com/MozOpenHard/B2G からの参照
→https://github.com/MozOpenHard/b2g-manifest に chirimen のXML 追加  
[Chirimen 用の Repository と XML の作成] 参照

## Chirimen 用の Repository と XML の作成

* [ ] Gaia のFork
* [ ] Gecko の Fork
* [ ] FringOne の Fork
* [ ] Gaia と Gecko のRevision を仮決め(Fork時点のV2.1の最新版で良いかなと。。)
* [ ] Chrimen の xml に 上記の Gaia と Gecko を追加(Revisionは上記のやつ)
→FringOneとの絡みをキチンと理解できていないので補足お願いします・・・

naobsd, 2015-04-20:
@YusukeYamamoto さん
mozilla-b2gからforkしてRK3066/CHIRIMEN固有の修正を入れていくのが理想形とは思いますが、まずは https://github.com/flingone/ のリポジトリを参照する形で一旦動くものが作れる状態にしようかと思います。

naobsd, 2015-04-22:
B2G, b2g-manifestリポジトリに最小限の変更をpushしました。

以下の作業が行える状態ですが、現状ではビルドされるものはカーネル等含め全てmatchstick用のものになります。

```
git clone git://github.com/MozOpenHard/B2G.git
cd B2G
./config.sh chirimen
./build.sh
```

引き続き、少なくともdevice/rockchip/rk3066, kernel/, u-boot/をCHIRIMEN用のものにする必要があります。

(ちなみにmatchstick用のイメージもCHIRIMENでとりあえず起動はします)

YusukeYamamoto, 2015-04-22:
@naobsd さん
ありがとうございます。今clone中です。

GaiaとGeckoは私の方でhttps://github.com/MozOpenHard/以下にPush + manifest更新しておきますね。
※少なくともGeckoはPushしておかないと、 @masap さんがGPIOの実装した奴の格納できないので・・・

naobsd, 2015-04-22:
upstreamからv2.1 branchをもってきて、v2.1はupstreamのmirrorとして残しておきつつ、mozopenhard-v2.1とか作ってそっち修正してく感じですかね

YusukeYamamoto, 2015-04-22:
そうですね。
v2.1 から mozopenhard-v2.1 の様にブランチ切って、修正分はそちらに反映。
あとは適宜 v2.1 を 最新に更新 → mozopenhard-v2.1 にも反映って感じの運用で良いかなぁと。

V2.1はコード変更も落ち着いてて、ほとんど改修が入ってないので、最新への追従も月1回で良さげです。

本家では毎月1回、月末の mergeday で必要なものを取り込んで  tag 追加しているので、そのタイミングで同期すれば良いかなと。

naobsd, 2015-04-22:
kernelとu-bootをMozOpenHardのものに切り替えました。といってもまだCHIRIMEN用のkernel configではなく、とりあえず起動する最小限のものですが。

明日あたりkernelとdeviceをCHIRIMEN用に調整して、私の方の「リポジトリを用意する」作業は一段落かなと思います。

YusukeYamamoto, 2015-04-22:
Gaia, Gecko の Forkと、ブランチの作成は完了しました。
CHIRIMEN用の xml のローカル編集完了で、本日お昼休みにビルドしてみて問題なければ commit します。
@naobsd さんの対応と私の対応をあわせて本 issue は完了しそうですね

YusukeYamamoto, 2015-04-23:
b2g-manifest 更新完了
それぞれ
https://github.com/MozOpenHard/gaia
https://github.com/MozOpenHard/gecko-dev
を参照するようにしてます。

本家のb2g-manifestではhttps://git.mozilla.org/releasesを参照しているので、こちらのものをベースにすべきか
github上のそれぞれを参照すべきか悩んだのですが、github上のを参照しています。

理由として、Github上にあがっているgecko, gaiaは上記の git.mozilla.orgのものとほぼ同一(ブランチ名が一部ことなる）であること、Github上でForkした方が本家側への参照がきちんとされ、Fork元として表示されるので、こちらの方がGithubとしての使い方にあっているかと思ったからです。

MozOpenHard上の gaia gecko それぞれに mozopenhard を付与したブランチがあるので、そちらを変更していけばとりあえずOKです。

naobsd, 2015-04-26:
kernel, deviceをCHIRIMEN用にし、manifest, config.shを切り替えました。
(ちょっと不思議な設定にしないとうまく動かない所があり、XXXが入っていますが、いずれそのうち...)

kernelの方はまだ最小限で、HDMI, OTG, HOSTくらいしか動作確認していません。各種ピンはもう少しkernelをいじらないと使えない気がしますが、それぞれ別のissueとして一つずつ対応していこうかと思います。

naobsd, 2015-04-26:
ひとまずcloseとします

closed 2015-04-26 by naobsd