MozOpenHard/CHIRIMEN #3: Web GPIOの実装

naokisekiguchi, 2015-04-20:


masap, 2015-04-24:
@naobsd さん、 GPIO はシルク上には 6 という番号が振ってありますが、ファイルにマップされた場合も番号は 6 でしょうか？それとも他の番号になるでしょうか？

naobsd, 2015-04-24:
@masap 印刷とは異なる番号です。

facebookのファイルの所に「OINARI(仮) pin config.pdf」があります。一つ前のボード用で片方だけしか書いてないので参考程度ですが、ご確認ください。

masap, 2015-04-25:
@naobsd ありがとうございます！

naobsd, 2015-04-27:
今日明日くらいを目標に、とりあえず全部GPIOにするkernel作って動作確認と、/sys/class/gpio/の番号の対応表の作成をやろうかと思います
(というのはこの ISSUE でいいんですかね、微妙に)

masap, 2015-04-27:
ありがとうございます。とりあえず今のちりめんでポートは見えてます。

masap, 2015-04-27:
@naobsd 

> とりあえず全部GPIOにする

I2C もやりたいので、 I2C として使えるポートもあると助かります。

naobsd, 2015-04-30:
ひとまずできるだけGPIOとして使えるconfigにしました
https://github.com/MozOpenHard/linux-rockchip/commit/1d5adccaf3cea93dfbbe1713551b91605cf7a696

対応表
https://docs.google.com/spreadsheets/d/15oXFBj-NiUhpJR2bSo39Gobp3cEvYrNl7SsFuJfsQwc/edit?usp=sharing

GPIOにしていないピン
* PWM0
  * PWMのテスト用
  * とりあえずバックライトとして操作できる状態
* I2C0_SCL, I2C0_SDA
  * audio codecがI2C0に接続されている
* PMIC_PWRON
  * 回路的に入力としてしか使えない(と思われる)
  * とりあえずPOWERキーになっている

naobsd, 2015-04-30:
@masap とりあえず全部GPIOにするというのは、作業の段階としてという事で、最終的な仕様として全部GPIOでという意味ではないです

masap, 2015-05-04:
/sys/class/gpio 以下にアクセスすると Permission Denied になってしまいます。

1) /sys/class/gpio のアクセス権を変更することは可能でしょうか？
2) b2g プロセスを root 権限で起動することは可能でしょうか？

1) については、 /sys/class/gpio 以下の Permission を事前にフルアクセスに変更したとしても、
gpio ポートを export してできたファイルは root でのみアクセス可能なので、同じく Permission Denied になります。
なので、ドライバがファイルを生成する時点でのアクセス権を変更する必要があります。

2) については、 /system/b2g/b2g に setuid してみたのですが、結果は同じでした。

何か良い方法はあるでしょうか？

naobsd, 2015-05-05:
@masap さん
ひとまず init.rk30board.rc と ueventd.rk30board.rc を手で修正した boot.img を以下に置きました。
http://dl.linux-rockchip.org/mozilla/boot-gpio-i2c.img

/sys/class/gpio/export と export した gpio の direction, value の permission が 0666 になるはずです。
export を 0666 にしてしまうのはあまり安全ではない気がしますが、 Firefox OS アプリが直接触るわけでなければまぁいいのかなとも思います。

Firefox OS のアーキテクチャをちゃんと理解していないんですが、 b2g プロセスはアプリ毎の uid で動くんでしょうか。 b2g を root で動かすのはよろしくないのではと思います。

masap, 2015-05-05:
@naobsd さん

> ひとまず init.rk30board.rc と ueventd.rk30board.rc を手で修正した boot.img を以下に置きました。

ありがとうございます。これでやってみます！

> b2g を root で動かすのはよろしくないのではと思います。

はい、私も良くないと思います。あくまで正式な方法が見つかるまでの暫定処置と考えています。

masap, 2015-05-05:
@naobsd さん

export は 666 ですが、 direction と value は以下のようになっており、 w 権限が付与されていないようです。
また、可能であれば export と同様に unexport も 666 にしていただけますでしょうか。

-rw-r--r-- root     root         4096 2015-05-05 05:46 active_low
-rw-r--r-- root     root         4096 2015-05-05 05:45 direction
-rw-r--r-- root     root         4096 2015-05-05 05:46 edge
drwxr-xr-x root     root              2015-05-05 05:46 power
lrwxrwxrwx root     root              2015-05-05 05:46 subsystem -> ../../../../class/gpio
-rw-r--r-- root     root         4096 2015-05-05 05:46 uevent
-rw-r--r-- root     root         4096 2015-05-05 05:46 value

naobsd, 2015-05-05:
起動してから何十秒か経つと設定が反映されなくなるようですね...
http://dl.linux-rockchip.org/mozilla/boot-gpio-i2c.img
を workaround を入れたもので置き換えました。

ところでアプリ (Web API) レベルで export/unexport の操作ができる (する必要がある) のでしょうか ?

masap, 2015-05-05:
ありがとうございます。

export 後すぐに direction に書き込もうとすると Permission Denied になってしまうので、 Gecko 側においても usleep(10000) を入れることで回避できました。 #14 にも追記しておきます。

masap, 2015-05-05:
> ところでアプリ (Web API) レベルで export/unexport の操作ができる (する必要がある) のでしょうか ?

ユーザーから見たら export/unexport しなくても常時使えるようにしておくのが良いという意見はありますね。
使用可能な GPIO 番号一覧を返す API が欲しいという意見もあるので、いっしょにやるかも知れません。

naobsd, 2015-05-06:
あまり深く考えていないのですが、

1. アプリが使えるGPIO番号 (可変)
2. (他の)アプリが使用中のGPIO番号 (可変)
3. システム(ボード)としてアプリに使用させてよいGPIO番号 (固定)

くらいの概念があって(1+2=3)、 sysfs での export/unexport は 3. の為に必要な Linux 環境固有の操作だと思っています。 Web GPIO が「Linux の」 GPIO を操作する API でないなら、 export/unexport という操作はそもそもアプリから見えるところにあるものではないと思います。

CHIRIMEN では CN2-15 (GPIO6_A1/gpio353) は常に GPIO と決まっていますので、 export 自体は起動時に常に行っておくようにできます。そうすれば gecko での usleep(10000) は必要なくなります。

他の「GPIO としても使える」ピンについては、 CHIRIMEN においては必ずしも export すれば GPIO として使える(使ってよい)ものではなく、(現状では)他のドライバがそのピンを使っていない状態にしておく (kernel config を調整しておく) 必要があります。

(実際には、他の function で使用中の pin を export できてしまう場合がありますが、システムとしては正しい状態ではないので、どうなるか保証できません)

----

起動中に GPIO とその他の function を切り替えられるシステム・ボードについては(今のところ私は)考慮していませんが、もしかしたら Pin configuration の為の (Web GPIO とは別の) API が必要かもしれませんね。 export/unexport 操作は、こちらに属するものになると思います。

(CHIRIMEN も起動中に function を切り替えられるようにできるかもしれませんが、カーネルの修正が必要です)

masap, 2015-05-06:
検討ありがとうございます。

[export/unexport]
ある特定のピンを、起動時に GPIO から I2C へ切り替えるような運用は私の方も想定していません。
また、 export は GPIO として既に定義されているピンに対してしか行えない旨も理解しています。
それ故、GPIO として定義した時点で export もしておくべきという意見も妥当であると思います。

[gecko での usleep(10000)]
direction については、動的である必要は無いにせよ、ユーザーが設定できるようにすべきと思っています。 
direction を切り替えた際に、 value を open しようとすると、こちらも Permission Denied になるため usleep(10000) が必要になります。

naobsd, 2015-05-06:
@masap さん

私は direction については特にどうすべきという意見はなく、 value 同様に自由に変更できてよいものという認識です。 https://github.com/MozOpenHard/CHIRIMEN/issues/3#issuecomment-99287603 で、 export を見せる必要はない、 gecko での usleep(10000) が不要になる、というのはどちらも export についての事です。

https://github.com/MozOpenHard/CHIRIMEN/issues/3#issuecomment-99058756 では export -> direction 間で usleep(10000) が必要という事だけでしたが、本当に direction -> value 間も usleep(10000) が必要なのでしょうか ?

私が入れた workaround の影響で確かに chown/chmod が行われるタイミングは遅くなるのですが、 export 後十分時間が経った後であれば direction も value も permission は変更されているはずで、「direction を変えた後 permission が変更されるのを待つ」必要はないと思っているのですが...

masap, 2015-05-06:
@naobsd 

> #3 (comment) で、 export を見せる必要はない、 gecko での usleep(10000) が不要になる、というのはどちらも export に関する操作の事です。

はい。それについては理解しています。 direction はあくまで派生の話題として書きました。

> 本当に direction -> value 間も usleep(10000) が必要なのでしょうか ?

実際に試してみたところそうなっています。 export -> direction だけでなく direction -> value においても usleep(10000) が無いと Permission Denied になります。

masap, 2015-05-09:
ひとまず完成したと思うのでアップします。

1) こちらの system.img を解凍して焼き、
http://masap.sakura.ne.jp/firefoxos/system.img.zip

2) 深海さんに作成していただいた boot.img を焼き、
http://dl.linux-rockchip.org/mozilla/boot-gpio-i2c.img

3) このアプリを WebIDE をインストールして使用します。
http://masap.sakura.ne.jp/firefoxos/test-gpio.tar.gz

API はこちらの仕様をベースに export/unexport を削除する等したものです。
https://github.com/browserobo/WebGPIO/blob/master/index.html

naobsd, 2015-05-10:
現状のwork aroundを突っ込んでしまうか、別の方法を試してみるか、決めないといけませんね...

naokisekiguchi, 2015-05-11:
http://dl.linux-rockchip.org/mozilla/chirimen-test.zip
が焼いてあるCHIRIMENに以下のコマンドでイメージを書き込んだ後、再起動すると、linuxのペンギンが２つ表示された状態で止まってしまうのですが、何か手順が違っているでしょうか？

$ upgrade_tool di -s system.img
Download image ok.
$ upgrade_tool di -b boot-gpio-i2c.img
Download image ok.
この後再起動

naobsd, 2015-05-11:
@naokisekiguchi さん

chirimen-test.zip は一番最初の Android 4.2 ベースのもので、現状は mozopenhard-20150426.zip の Android 4.4 ベースのものを使っています。

kernel.img を書き換えれば動くような気もしますが、まずは mozopenhard-20150426.zip を一通り書き込んで動作確認をしてみてください。

(misc.img を入れ忘れた気もしますが、とりあえず起動はするかと...)

naokisekiguchi, 2015-05-11:
mozopenhard-20150426.zip内のkernel.imgを焼いたらとりあえず起動しました。ありがとうございます。

naokisekiguchi, 2015-05-18:
@masap さん
上記のイメージにはGPIOChangeEventのあたりも実装されていますか？

naobsd, 2015-05-18:
work weekの為にup-to-dateなfull imageを用意しなきゃいけないかな、と思っている間にもう前日になってしまいましたが... あった方がいいですよね ?

現状の work around はどう考えてもまっとうなやり方ではないのですが、とりあえず一旦 commit して build しようかと思います...

masap, 2015-05-18:
@naokisekiguchi change event は未実装です。

naokisekiguchi, 2015-05-20:
https://github.com/MozOpenHard/CHIRIMEN/issues/3#issuecomment-100440442
こちらのイメージで動作確認したところ、readの値がおかしいです。

タクトスイッチをつないでgpioの値を読み取ったところ、1163674816という値が出てきます。
adbでログインして/sys/class/gpio/gpio199/valueの値を読み取るとスイッチの操作で0か1が読み取れています。

naobsd, 2015-05-21:
boot-gpio-i2c.img に行った修正は commit/push しましたので、repo sync & build.sh で作れるはずです。(ビルド中)

masap, 2015-05-21:
@naokisekiguchi さん

read の不具合を修正しました。下記２ファイルを修正しました。

http://masap.sakura.ne.jp/firefoxos/system.img.zip
http://masap.sakura.ne.jp/firefoxos/test-gpio.tar.gz

naobsd, 2015-05-22:
gecko-dev の b2g34_v2_1-mozopenhard に https://github.com/MozOpenHard/gecko-dev/commit/673ead4e8b57a270cc56209fb14919ee4ccf5b78 **だけ** cherry-pick して build が通る事は確認しました。
(一つ前の "Patched files." は b2g34_v2_1-mozopenhard には merge しないようお願いします)

naobsd, 2015-05-23:
20150523 のイメージでAPIの動作確認をお願いします

masap, 2015-05-23:
マージありがとうございます。動作確認 OK です。 198(out)/199(in) にて確認しました。

masap, 2015-05-25:
サンプルコードをこちらに上げました。

https://github.com/MozOpenHard/examples/tree/master/test-gpio

masap, 2015-05-26:
実装完了のため close。

closed 2015-05-26 by masap

autonome, 2015-05-26:
\o/