MozOpenHard/CHIRIMEN #4: Web I2Cの実装

naokisekiguchi, 2015-04-20:


naobsd, 2015-04-30:
I2C0, I2C2ピンが使えるconfigです
https://github.com/MozOpenHard/linux-rockchip/commit/882dffaf013ddb4e961a4b6d32cb0f25c5daaac4

(I2C2ピンはGPIOとしては動作する事を確認済、I2Cデバイスがない為I2Cとしての動作は未確認)

naobsd, 2015-05-05:
/dev/i2c* の permission を 0666 にする修正を行った boot.img を以下に置きました。
http://dl.linux-rockchip.org/mozilla/boot-gpio-i2c.img

naobsd, 2015-05-05:
i2c-tools 入れといた方がよいでしょうか ?

masap, 2015-05-05:
> /dev/i2c* の permission を 0666 にする修正を行った boot.img を以下に置きました。

ありがとうございます！

> i2c-tools 入れといた方がよいでしょうか ?

ioctl 経由で叩くので、とりあえず無くても大丈夫です。

naokisekiguchi, 2015-05-14:
@masap さん
ioctlでi2cを叩くコードをお持ちだったりしますか？
http://blog.psyche.gaso.jp/20131013-1.html
このあたりを参考にi2cデバイスを叩こうとしているのですが。
android-ndkでのビルドとCHIRIMEN上での実行までは行いましたが、ファイルディスクリプタのopenのところでエラーが発生しています。
/dev/以下にはi2c-*はなく(i2c-detectというのはありますが)、/sys/bus/i2c/devices/以下にi2c-0などファイルがあるのですが、こちらを指定すればいいのでしょうか？

mozopenhard-20150426.zipに対してhttp://dl.linux-rockchip.org/mozilla/boot-gpio-i2c.imgを焼いたものを使用しています。

@naobsd さん
i2c-tools入れていただけると私は助かります(^^;
以前のイメージでi2c-toolsを入れてもらったものがありますが、そちらのバイナリをコピーしても大丈夫ですか？

naobsd, 2015-05-14:
@naokisekiguchi さん
i2cを使うには https://github.com/MozOpenHard/linux-rockchip/commit/882dffaf013ddb4e961a4b6d32cb0f25c5daaac4 のソースで kernel.img を作ってください。
イメージにi2c-toolsも入れるようにしますが、とりあえず古いイメージのが動くと思います。

masap, 2015-05-15:
I2C0 はオーディオコーデックに接続とのことなので、コーデックを壊すと嫌なので I2C2 に接続して試してみました。

デバイスの open はできますが、 ioctl で書き込みを行なおうとすると Try again (errno=11) でエラーになります。
dmesg には下記が表示されます。下記エラー時は /dev/i2c-4 を使用していますが i2c-0, 1, 2 でも同様です。 i2c-3 は存在しません。
また、 slave address 0x3e ですが、こちらは Raspberry Pi で動作実績のあるアドレスですので間違いはないです。
もうちょい調べてみます。

<3>[  183.746955] i2c i2c-4: No ack, Maybe slave(addr: 0x3e) not exist or abnormal power-on, retry 2...
<3>[  183.747304] i2c i2c-4: No ack, Maybe slave(addr: 0x3e) not exist or abnormal power-on, retry 1...
<3>[  183.747643] i2c i2c-4: No ack, Maybe slave(addr: 0x3e) not exist or abnormal power-on, retry 0...

naobsd, 2015-05-15:
念の為、外に出ているのはi2c0/2だけで、i2c1/4はボード上のpmic/rtcとeepromに繋がっているだけで外には出ていません。

なぜエラーになるかについては、ちょっと今はわかりませんね...

naobsd, 2015-05-18:
@masap さん

I2C2のSCL/SDAに何かしら信号が出ているか見れますでしょうか？
なんとなく、I2C0だと動くんじゃないかという気が...

naokisekiguchi, 2015-05-18:
このセンサを使って、お手製のプログラムを実行したらそれっぽい値が取れています。（/dev/i2c-0使用）
http://akizukidenshi.com/catalog/g/gM-06675/

i2ctest.c
#include <string.h>
#include <linux/i2c.h>
#include <sys/ioctl.h>
#include <fcntl.h>
#include <stdio.h>

#define DATA_NUM 10

double get_temp(int rtc){
        char buf[4];
        double temp;

        if (read(rtc, buf, 4) != 4) {buf[0] = 0x00; buf[1] = 0x00;}
        temp = ((double)((int)buf[0] << 5) + ((int)buf[1] >> 3)) / 16; //13bitモード
        //temp = ((double)((int)buf[0] << 8) + (int)buf[1]) / 128; //16bitモード
        return temp;
}

int main(void){
        int fd, rtc, i, count;
        if ((rtc = open("/dev/i2c-0", O_RDWR)) < 0){
                printf("[Error] open i2c port\n");
                return(-1);
        }
        if(ioctl(rtc, I2C_SLAVE, 0x48) < 0){
                printf("[Error] access to 0x48\n");
                return(-1);
        }

	printf("temp = %f",get_temp(rtc));
        return(0);
}

MozOpenHard/B2Gのレポジトリを使ってconfig,buildしたイメージを使っています。 MozOpenHard/linux-rockchip@882dffaの設定が反映されていたようなので。

昔のイメージからi2ctoolsのバイナリをコピーして実行してみた結果です。
i2cdetectでセンサのアドレスがうまく見えないのは原因がわかりませんが、i2cgetではそれっぽい値が取れています。

# i2cdetect -y 0                                     
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 
10: 10 11 12 13 14 15 16 17 18 19 UU 1b 1c 1d 1e 1f 
20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 
30: -- -- -- -- -- -- -- -- 38 39 3a 3b 3c 3d 3e 3f 
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 
70: 70 71 72 73 74 75 76 77 
# i2cget -y 0 0x48 0x00 w                            
0x900f

naobsd, 2015-05-19:
@naokisekiguchi さん、確認ありがとうございます。

その環境で接続だけを I2C2 に変更した場合、どうなるでしょうか?(動かないのではないでしょうか)

naobsd, 2015-05-21:
boot-gpio-i2c.img に行った修正は commit/push しましたので、repo sync & build.sh で作れるはずです。(ビルド中)

naobsd, 2015-05-21:
i2c-toolsを作るようにしました

naokisekiguchi, 2015-05-21:
@naobsd さん
上記、同じ環境でi2c2につないでみても値は読み取れていました。

    # i2cdetect -y 2                                    
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
    00:          03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 
    10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 
    20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 
    30: -- -- -- -- -- -- -- -- 38 39 3a 3b 3c 3d 3e 3f 
    40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 
    50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
    60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 
    70: 70 71 72 73 74 75 76 77                         
    # i2cget -y 2 0x48 0x00 w                            
    0x780d

naobsd, 2015-05-21:
@naokisekiguchi さん
確認ありがとうございます！

@masap さん
動かないのはどのようなデバイスで、どのようなアクセスをしているのでしょうか？
デバイスまたはアクセス方法によってダメという可能性もあると思いますので、ダメなようでしたら新しいissueとして具体的な情報をあげておいてもらえると助かります。

masap, 2015-05-22:
すいません。 i2c0, i2c2 共に動作しました。配線ミスってるだけでした。

masap, 2015-05-23:
I2C のコードも GPIO と同じブランチに push しました。

ビルド済みのバイナリはこちらです。
http://masap.sakura.ne.jp/firefoxos/system.img.zip

この LCD (https://www.switch-science.com/catalog/1405/) を動かすサンプルコードは下記に置きました。
http://masap.sakura.ne.jp/firefoxos/test-i2c.tar.gz

この LCD でテストできるのは 8bit の書き込みのみなので、下記の機能を持つデバイスをお持ちの方はテストしていただけると助かります。

16bit 書き込み
8bit 読み取り
16bit 読み取り

naobsd, 2015-05-23:
[mozopenhard_20150523.zip](http://dl.linux-rockchip.org/mozilla/) のイメージでAPIの動作確認をお願いします

masap, 2015-05-23:
マージありがとうございます。動作確認 OK です。 8bit の書き込みにて確認し、 i2c0, i2c2 共に OK でした。

masap, 2015-05-25:
サンプルコードをこちらに上げました。

https://github.com/MozOpenHard/examples/tree/master/test-i2c

naokisekiguchi, 2015-05-26:
@masap　さん
readをしてみたいと思うのですが、いまいち使い方がわかりません。
https://github.com/MozOpenHard/CHIRIMEN/issues/4#issuecomment-103031635
例えば、こちらのコメントのセンサ、プログラムと同じことやろうと思うと、どういったコマンドになりますか？

# 手元に同じLCDがあったので動作確認したら無事動きました。

masap, 2015-05-26:
@naokisekiguchi 

そちらでも動いたとのこと良かったです。

> readをしてみたいと思うのですが、いまいち使い方がわかりません。

navigator.mozI2c.open の第二引数に 0x48 を渡すように変更し、メイン部分を以下のように書くとどうでしょうか。

```javascript
  navigator.requestI2CAccess().then(
    function(i2cAccess) {
      var ports = i2cAccess.ports;
      var port = ports.get(2);
      console.log(port.read(0x00, false));
    },
    function(error) {
      console.log(error.message);
    }
  );
```

naokisekiguchi, 2015-05-26:
@masap さん
数値は帰って来るのですが。
readコマンドのcommand,isOctet
writeコマンドのcommand,value
はそれぞれどういう意味ですか？

masap, 2015-05-26:
@naokisekiguchi さん

数値は返ってくるものの、値がおかしいということでしょうか？

read/write 共に command というのは、読み取り/書き込み対象のアドレスです。

read コマンドの isOctet が true の場合は 1バイトの読み取り、 false の場合は 2 バイトの読み取りです。
該当ハードのマニュアルを見ると 13bit/16bit とあるので 2 バイト読み取りにしています。

write コマンドの value は書き込む値です。

明日 (05/27) 19:30 に別件で Mozilla に行くので 17:00 とか 18:00 に機材使いながら共同デバッグ等しますか？

masap, 2015-05-26:
チップのデータシートを読むと
http://www.analog.com/media/en/technical-documentation/data-sheets/ADT7410.pdf

16bit 読み取りじゃなくて 8 bit 読み取りかもです。

```javascript
  navigator.requestI2CAccess().then(
    function(i2cAccess) {
      var ports = i2cAccess.ports;
      var port = ports.get(2);
      console.log(console.log(port.read(0x00, true) << 8 + port.read(0x01, true)););
    }
  );
```

naokisekiguchi, 2015-05-26:
8bitずつ読み取りにしてみたらそれっぽい値が取れました。       

    Promise.all([
      port.read(0x00,true),
      port.read(0x01,true)
    ]).then(function(v){
      var temp = (v[0] << 8) + v[1];
      console.log(temp / 128.0 );
    });

naokisekiguchi, 2015-05-26:
#3, #4, 共に一旦実装はできてコードがプッシュされたのでcloseして、今後のバグは個別にissue切りましょうか？

masap, 2015-05-26:
それで良いと思います。

masap, 2015-05-26:
実装完了のため close。

masap, 2015-07-11:
At the bug https://bugzilla.mozilla.org/show_bug.cgi?id=1166634, @naokisekiguchi suggested to implement multi device support.

So I made the patch and tested it.

Now I want to push it.
@naobsd I tried to push by this command according to your previous comment.
But it filed with below messages.
How could I push this ?

$ git push git@github.com:MozOpenHard/gecko-dev.git webgpio:b2g34_v2_1-webgpio
To git@github.com:MozOpenHard/gecko-dev.git
 ! [rejected]        webgpio -> b2g34_v2_1-webgpio (non-fast-forward)
error: failed to push some refs to 'git@github.com:MozOpenHard/gecko-dev.git'
hint: Updates were rejected because a pushed branch tip is behind its remote
hint: counterpart. Check out this branch and integrate the remote changes
hint: (e.g. 'git pull ...') before pushing again.
hint: See the 'Note about fast-forwards' in 'git push --help' for details.

naobsd, 2015-07-11:
@masap which branch/commit is the base of your new change?
if you are not sure, just try

    $ git push git@github.com:MozOpenHard/gecko-dev.git webgpio:b2g34_v2_1-webi2c

masap, 2015-07-11:
@naobsd 

> which branch/commit is the base of your new change?

The base commit is 6bc3525f524d267220575af20ce2d5d9caffa8d1 of Gecko.
My branch is based on the commit and named "webgpio".

> $ git push git@github.com:MozOpenHard/gecko-dev.git webgpio:b2g34_v2_1-webi2c

Thank you ! It works.
Could you cherry-pick this commit (ec05d2648d70e291a5c36e3dfee286174d9f580a) ?

naobsd, 2015-07-12:
I pushed that commit into b2g34_v2_1-mozopenhard.

closed 2015-07-12 by masap