MozOpenHard/CHIRIMEN #7: ドキュメントの用意

naokisekiguchi, 2015-04-20:
MDN上にCHIRIMENに関するドキュメントを用意する。

dynamis, 2015-04-20:
スマホは Phone Guide ディレクトリの配下に各端末がある:
https://developer.mozilla.org/ja/Firefox_OS/Phone_guide

Board Guide ページを作ってその下に CHIRIMEN ページを作成、Phone Guide から Board Guide へのリンクをするとかいかがでしょう？

https://developer.mozilla.org/ja/Firefox_OS/Board_guide
https://developer.mozilla.org/ja/Firefox_OS/Board_guide/CHIRIMEN

それぞれ親のページで「サブページの作成」をすれば作れるはずです。

YusukeYamamoto, 2015-04-22:
取りあえず
https://developer.mozilla.org/ja/Firefox_OS/Board_guide
https://developer.mozilla.org/en-US/Firefox_OS/Board_guide
は作成。
先にjaを作成したところ、en-USへの翻訳リンクが無く翻訳できない状態に・・・
しかたがないので、en-USのURLで新規ページの作成をし、翻訳を行ったが、
今度は
Document with this Slug and Locale already exists.
と表示され、保存できない事態に・・・

なので、
https://developer.mozilla.org/ja/Firefox_OS/Board_guide
https://developer.mozilla.org/en-US/Firefox_OS/Board_guide
の両者は存在するが、相互に翻訳の為のリンク?ができていない。

一旦 ja を削除してから en-US を翻訳しようとしたが、そもそもでページの削除も出来ず。。。

@dynamis さん、 @dadaa さん たすけてください・・

dynamis, 2015-04-23:
おおぅ。ja への tranlsation 追加はできなくなっているのですね。
対訳関係のリンク追加をお願いしてみましょう。最悪 ja を削除して翻訳として再作成するばあいでもそれは結局 Admin リクエストが必要ですし。

dynamis, 2015-04-24:
bug: https://bugzilla.mozilla.org/show_bug.cgi?id=1158001

メールでも問い合わせはしておいた。取り敢えず翻訳リンク関係無しのままでも気にせず追加して行って頂ければ良いかと。

naobsd, 2015-04-27:
直っているようですね

dynamis, 2015-04-27:
https://bugzilla.mozilla.org/show_bug.cgi?id=1158001 で書き込まれているとおり、ja ページの編集画面から「ページのタイトルとプロパティの編集」をクリックして「Find the translation source with the lookup below and then click "Save Changes."」というフィールドで翻訳元 en-US としたいページを選択したらリンク関係は作れるとのことです。

この場合でも ja がオリジナルではなく en-US がオリジナルという扱いになるとい課題はありますが、取り敢えずユーザ目線で言語間リンクは問題なく出来そうですので (MDN が ja オリジナルとかに対応できる夢の未来が到来しない限り) 当面この方法でやっていきましょう。

YusukeYamamoto, 2015-04-27:
> ja ページの編集画面から「ページのタイトルとプロパティの編集」をクリックして「Find the translation source with the lookup below and then click "Save Changes."」というフィールドで翻訳元 en-US としたいページを選択したらリンク関係は作れるとのことです。

なるほど、確かに翻訳元のページの指定ができるんですね・・・
※日本語ページが独立してあった際に選択できたか覚えてませんが・・・

とはいえ、次から作成するときは en-US で作成して、そのあと翻訳の方が良いですね。
CHIRIMEN ページの作成しますね

YusukeYamamoto, 2015-04-27:
CHIRIMENページに書くことメモ

* CHIRIMEN の概要説明
  - CHIRIMEN とは何ぞや
   →いまは手に入らないよーという旨も忘れずに
  - リンクは Github, Chirimen 公式, Google Group, facebookも?
* How to build 
 - バイナリから焼く方法
   - Linux, Windows, Mac(可能なら)
 - ビルド方法
   - Linux, Windows, Mac(可能なら)
 - 他のデバイスに焼いてお試しする方法も?
  - 書いて良いのか知らないけど OWBとかも?

他にあればコメントください

closed 2016-07-05 by MSakamaki