MozOpenHard/CHIRIMEN #11: Mac OS X 環境でのファームウェア更新

naobsd, 2015-04-22:
ブートローダの更新は Rockchip 製の upgrade_tool (Linux x86) や AndroidTool (Windows) で行う必要がある為、 Mac OS X では (VM 内で Linux/Windows を使用しない限り) 更新できない。

しかしブートローダ以外のイメージ (boot.img, system.img 等) については、 Mac OS X 環境でも更新できるはず。

* rkflashtool https://github.com/linux-rockchip/rkflashtool
** libusb があればビルドして使えるはず (誰かに試してほしい)
* fastboot
** ブートローダが U-Boot であれば (& バグを直せば) 使えるはず

#10 SD カードにインストールできれば何も問題はない

YusukeYamamoto, 2015-04-27:
> rkflashtool

libusb を入れた後、Mac OS X 10.9.2 (marvericks) では make -> バイナリ が出来ました。
焼くのは後ほどやります。
※申し訳ないですが、GW入ってからの予定です。

naokisekiguchi, 2015-05-14:
mac上のVM内にubuntu を入れて焼いているのですが、
・CHIRIMENをVM側に接続してupgrade_toolで焼く
・wait for loader device （うろ覚え）のような表示の際にもう一度CHIEIMENをVM側に接続する
という形で焼いています。
イメージを焼く過程で一度CHIRIMENの認識がHOST PC側に移ってしまっているようで。

本質的な解決策でないですが、参考まで。

naokisekiguchi, 2015-05-14:
wait for loader startでした

naobsd, 2015-05-14:
upgrade_toolでloaderを焼く際にリセットがかかる(デバイスの抜き差しと同じ状態になる)ので、VMの中でupgrade_toolを使う場合、guestに自動的に見せるような設定がなければもう一度guestに見せる作業が必要になると思います。

closed 2016-07-05 by MSakamaki