MozOpenHard/CHIRIMEN #13: 書き込み用イメージの作成

naobsd, 2015-04-26:
after ./build.sh, images for flashing need to be prepared as follows

* out/target/product/chirimen/kernel
  * rkcrc -k kernel path/to/Image/kernel.img
* out/target/product/chirimen/ramdisk.img
  * rkcrc -k ramdisk.img path/to/Image/boot.img
* out/target/product/chirimen/recovery.img
  * cp recovery.img path/to/Image/recovery.img
* out/target/product/chirimen/system.img
  * out/host/linux-x86/bin/simg2img system.img tmp.img
  * resize2fs -M tmp.img
  * cp tmp.img path/to/Image/system.img

these commands should be run in flash.sh.

naobsd, 2015-04-27:
commands for flashing:
    $ upgrade_tool di -k kernel.img
    $ upgrade_tool di -b boot.img
    $ upgrade_tool di -r recovery.img
    $ upgrade_tool di -s system.img

naokisekiguchi, 2015-05-18:
Preparing 'rkcrc' command. 

http://linux-rockchip.info/mw/index.php?title=Building_Kernel_for_Linux_Distribution
$ git clone --depth 1 https://github.com/naobsd/rkutils
$ cd rkutils
$ gcc rkcrc.c -o rkcrc

Is it OK?

naobsd, 2015-05-19:
@naokisekiguchi yes, but

    $ git clone https://github.com/naobsd/rkutils
    $ cd rkutils
    $ make rkcrc

should be more generic.

(btw that wiki contains a lot of errors, please do not rely that so much...)

closed 2016-07-05 by MSakamaki