MozOpenHard/CHIRIMEN #14: ueventd.rc rules are not applied under certain conditions

naobsd, 2015-05-05:
following rules are not applied under certain conditions.

    /sys/devices/virtual/gpio/gpio*   direction   0666   root       root
    /sys/devices/virtual/gpio/gpio*   value       0666   root       root

workaround: adding `usleep(10000);` just before chown()/chmod() in system/core/init/devices.c:fixup_sys_perms()
(probably this is not correct fix...)

masap, 2015-05-05:
This delay is required on gecko side also.

I have inserted ``` usleep(10000); ``` between exporting and setting direction.

naobsd, 2015-05-21:
workaround is commited https://github.com/MozOpenHard/platform_system_core/commit/4f177d0b25c259a9340329db0cea01c8feabce8e

closed 2016-07-05 by MSakamaki