MozOpenHard/CHIRIMEN #17: build config failed: chirimen.xml missing

autonome, 2015-05-20:
I have followed the build steps listed at: https://etherpad.mozilla.org/wot-ww

For "config.sh chirimen" I receive the error:

fatal: manifest 'chirimen.xml' not available
fatal: manifest chirimen.xml not found

autonome, 2015-05-20:
@naobsd Hello, any idea about this problem?

naobsd, 2015-05-20:
@autonome  I cannot reproduce problem.

please run "git branch" and "git remote -v" under B2G directory, then tell me result.

autonome, 2015-05-20:
dietrich@firefoxos:~/chirimen$ git branch
* mozopenhard
dietrich@firefoxos:~/chirimen$ git remote -v
origin    git@github.com:MozOpenHard/B2G.git (fetch)
origin    git@github.com:MozOpenHard/B2G.git (push)

naobsd, 2015-05-20:
hmm... how about

    $ cd .repo/manifests
    $ git branch
    $ git remote -v
    $ ls

autonome, 2015-05-20:
dietrich@firefoxos:~/chirimen/.repo/manifests$ git branch
* default
dietrich@firefoxos:~/chirimen/.repo/manifests$ git remote -v
origin    git://github.com/MozOpenHard/b2g-manifest (fetch)
origin    git://github.com/MozOpenHard/b2g-manifest (push)

naobsd, 2015-05-20:
"ls" result is missing...

btw, when "fatal: manifest 'chirimen.xml' not available" is displayed?
I thought it's displayed immediately just after "config.sh chirimen". should I wait end of download?

autonome, 2015-05-20:
Error is displayed immediately after ./config.sh chirimen.

lrwxrwxrwx 1 dietrich dietrich 11 May 19 23:23 aries.xml -> shinano.xml
-rw-rw-r-- 1 dietrich dietrich 9111 May 19 23:23 base-caf-jb.xml -rw-rw-r--
1 dietrich dietrich 10025 May 19 23:23 base-jb.xml -rw-rw-r-- 1 dietrich
dietrich 359 May 19 23:23 base-kk-aosp.xml -rw-rw-r-- 1 dietrich dietrich
763 May 19 23:23 base-kk-caf.xml -rw-rw-r-- 1 dietrich dietrich 9595 May 19
23:23 base-kk.xml -rw-rw-r-- 1 dietrich dietrich 699 May 19 23:23
base-l-aosp.xml -rw-rw-r-- 1 dietrich dietrich 11268 May 19 23:23
base-l.xml lrwxrwxrwx 1 dietrich dietrich 12 May 19 23:23 default.xml ->
emulator.xml lrwxrwxrwx 1 dietrich dietrich 11 May 19 23:23 dolphin-512.xml
-> dolphin.xml -rw-rw-r-- 1 dietrich dietrich 1244 May 19 23:23 dolphin.xml
-rw-rw-r-- 1 dietrich dietrich 1222 May 19 23:23 emulator-jb.xml -rw-rw-r--
1 dietrich dietrich 925 May 19 23:23 emulator-kk.xml -rw-rw-r-- 1 dietrich
dietrich 932 May 19 23:23 emulator-l.xml lrwxrwxrwx 1 dietrich dietrich 15
May 19 23:23 emulator-x86-jb.xml -> emulator-jb.xml lrwxrwxrwx 1 dietrich
dietrich 15 May 19 23:23 emulator-x86-kk.xml -> emulator-kk.xml lrwxrwxrwx
1 dietrich dietrich 14 May 19 23:23 emulator-x86-l.xml -> emulator-l.xml
lrwxrwxrwx 1 dietrich dietrich 12 May 19 23:23 emulator-x86.xml ->
emulator.xml -rw-rw-r-- 1 dietrich dietrich 8285 May 19 23:23 emulator.xml
-rw-rw-r-- 1 dietrich dietrich 2781 May 19 23:23 flame-kk.xml -rw-rw-r-- 1
dietrich dietrich 3196 May 19 23:23 flame.xml -rw-rw-r-- 1 dietrich
dietrich 9797 May 19 23:23 flatfish.xml -rw-rw-r-- 1 dietrich dietrich 1439
May 19 23:23 flo.xml -rw-rw-r-- 1 dietrich dietrich 7337 May 19 23:23
galaxy-nexus.xml -rw-rw-r-- 1 dietrich dietrich 6858 May 19 23:23
galaxy-s2.xml -rw-rw-r-- 1 dietrich dietrich 8220 May 19 23:23 hamachi.xml
-rw-rw-r-- 1 dietrich dietrich 7510 May 19 23:23 helix.xml lrwxrwxrwx 1
dietrich dietrich 9 May 19 23:23 inari.xml -> otoro.xml -rw-rw-r-- 1
dietrich dietrich 11316 May 19 23:23 keon.xml -rw-rw-r-- 1 dietrich
dietrich 7697 May 19 23:23 leo.xml -rw-rw-r-- 1 dietrich dietrich 1654 May
19 23:23 nexus-4-kk.xml -rw-rw-r-- 1 dietrich dietrich 1573 May 19 23:23
nexus-4.xml -rw-rw-r-- 1 dietrich dietrich 1641 May 19 23:23 nexus-5-l.xml
-rw-rw-r-- 1 dietrich dietrich 1692 May 19 23:23 nexus-5.xml -rw-rw-r-- 1
dietrich dietrich 6976 May 19 23:23 nexus-s-4g.xml -rw-rw-r-- 1 dietrich
dietrich 6881 May 19 23:23 nexus-s.xml -rw-rw-r-- 1 dietrich dietrich 8131
May 19 23:23 otoro.xml -rw-rw-r-- 1 dietrich dietrich 7521 May 19 23:23
pandaboard.xml -rw-rw-r-- 1 dietrich dietrich 11494 May 19 23:23 peak.xml
-rw-rw-r-- 1 dietrich dietrich 622 May 19 23:23 README.md -rw-rw-r-- 1
dietrich dietrich 7556 May 19 23:23 rpi.xml -rw-rw-r-- 1 dietrich dietrich
2524 May 19 23:23 run_travis_tests.sh -rw-rw-r-- 1 dietrich dietrich 2909
May 19 23:23 shinano.xml -rw-rw-r-- 1 dietrich dietrich 7572 May 19 23:23
tarako.xml lrwxrwxrwx 1 dietrich dietrich 9 May 19 23:23 unagi.xml ->
otoro.xml -rw-rw-r-- 1 dietrich dietrich 9877 May 19 23:23 vixen.xml
-rw-rw-r-- 1 dietrich dietrich 7797 May 19 23:23 wasabi.xml

naobsd, 2015-05-20:
you should have something (env?) to specify (wrong) branch for manifest.
"default" branch for manifests must be same as "mozopenhard-v2.1" from https://github.com/MozOpenHard/b2g-manifest

autonome, 2015-05-20:
Yesssss I was in a screen session from a previous build from a different device!

My apologies, thanks for your help :)

closed 2015-05-20 by autonome