MozOpenHard/CHIRIMEN #20: support USB-ethernet adapter

naobsd, 2015-05-21:
support AX88772 based USB-ethernet adapter

naobsd, 2015-05-21:
kernel device driver is already enabled, but dhcpcd is not running by default. if I start dhcpcd manually, IP address is assigned, but browser doesn't recognize internet connection.

it needs more investigation...

autonome, 2015-05-21:
There may be some helpful information in these bugs:

https://bugzilla.mozilla.org/show_bug.cgi?id=1082969
https://bugzilla.mozilla.org/show_bug.cgi?id=1087548

naobsd, 2015-06-03:
@autonome thanks for the hint!

this is workaround,

1. add `user_pref("network.gonk.manage-offline-status", false);` to /data/b2g/mozilla/*.default/prefs.js
2. reboot
3. run following commands on root shell (serial console or adb shell with su)

   ```
netcfg eth0 dhcp
ndc resolver setdefaultif eth0
ndc resolver setifdns eth0 localhost 8.8.8.8
   ```

autonome, 2015-06-04:
1. Will you be adding this to the base build to make it supported out of
the box?

2. Is there a wiki page for collecting Chirimen tips like this?

closed 2016-07-05 by MSakamaki