MozOpenHard/CHIRIMEN #25: At least 2 ADC pins

djpnewton, 2015-07-19:
This is a big weakness of the raspberry pi and would be a good point of difference

arky, 2015-07-21:
+1 
More Analog to digital conversion(ADC) pins would be useful. The use case is connecting analog temperature,humidity, vibration sensors to Chirimen.

dynamis, 2015-11-26:
+1.

Current board don't have this and we're planning to include ADC as an optional starter kit.
We should consider to include ADC pins for next model.

naobsd, 2015-11-26:
1 ADC is available on current CHIRIMEN board (CN1-6). this issue is request for _2_ ADC, right?

----

random notes about ADC...

* number of ADC

RK3066 has 3 ADC. on CHIRIMEN, currently 1 ADC is exposed, 1 ADC is used for "Recover Mode" button, 1 ADC is unused.

technically, it's possible to assign GPIO for "Recover Mode" button. so, if we need, we can expose 3 ADC at most in the future(if it uses RK3066).

* low level API

currently there is no interface (/dev, /sys, etc) to read ADC from userland. it needs to be implemented before Web API. (low level API should be same as latest Linux)

* reference voltage

reference voltage of RK3066 ADC is 2.5V which is not exposed(as hole/pin for users) on the CHIRIMEN board.

some times ago I asked about it, and answer was "it's ok" without detail. I have no idea how it's ok, but it should be explained when it becomes user accessible. (or, is it common knowledge?)

dynamis, 2015-11-28:
Ahh, CHIRIMEN already have an ADC pin but we don't have low level API for it yet.

Sorry I didn't noticed about it. When I see this issue, I thought CHIRIMEN don't have any ADC pin. In that case, It don't think we need prioritize to have 2 or more ADC pins.

If we want to use multiple analog sensors, we can use I2C ADC board to use multiple (many) analog sensors. We can continue to focus on I2C to use multiple sensors (or servo etc).

closed 2016-07-05 by MSakamaki