MozOpenHard/CHIRIMEN #26: Chirimen Default Starter App

arky, 2015-08-29:
It would be a good idea to replace default 'Homescreen' app in the pre-built image with custom Chirimen default app. 

Here are few ideas for the default starter app: 
* Will provide a screen with some information about Chirimen. 
* It will run a basic blink example 
* An input UI element on the page will enable users to change the rate of blink (in seconds)
* Users who connected to Chirimen using the Firefox remote connection manager, could change the blink rate using Devtools console 
* If certain component (a gyro) is connected on pre-assigned GPIO ports then default app will use    Screen.orientation Web API.

autonome, 2015-08-30:
+1. The smartphone Homescreen is not very useful for IoT development case.
Probably still need a way to access some basics like the Settings app.

arky, 2015-09-09:
Most embedded hardware (ie.. OpenWRT, Arduino Yun) exposes the settings via a onboard web server. You just point your browser to particular port to access the settings UI.  

Here is how Yun does it: 
https://www.arduino.cc/en/Guide/ArduinoYun#toc14

dynamis, 2015-11-26:
Note about current workaround:

You can insert simple script to launch specific app in homescreen.

Please insert ```<script defer="" src="js/launchapp.js"></script>``` tag in index.html (at end of HEAD or BODY tag) of homescreen
https://github.com/mozilla-b2g/gaia/blob/master/apps/homescreen/index.html
and add apps/homescreen/js/launchapp.js file like this:

https://gist.github.com/dynamis/cb25a9b743a7a79401f4

If you want to launch your original app, please specify origin in your app manifest. Then the app id will be the origin, not random UUID and you can specify the app easily.

closed 2016-07-05 by MSakamaki