MozOpenHard/CHIRIMEN #27: Chirimen-Dev Mailing List

arky, 2015-08-29:
Please kindly create a chirimen development (chirimen-dev) mailing-list. This will improve communication and help remote contributors to collaborate.  It is really difficult to keep track of conversations on Facebook group. Thanks!

stretchyboy, 2015-08-30:
:+1:  But how will we handle translation?

naokisekiguchi, 2015-08-31:
:+1:  I had created a google group account named CHIRIMEN ( https://groups.google.com/forum/?hl=ja#!forum/chirimen ), but it's not used much. What settings are prefered? or should we create a new one?

arky, 2015-09-09:
Not sure @stretchyboy you meant by translation. I used this URI to access English UI of Google Groups 
https://groups.google.com/forum/?hl=en#!forum/chirimen

stretchyboy, 2015-09-09:
Many of the facebook group posts are in Japanese and Facebook sometimes
offers an automatic translation, which is helpful. I was wondering if
google groups includes a similar (or better ) feature.

YusukeYamamoto, 2016-01-18:
@dadaa, @dynamis, @naokisekiguchi  : It might be think that we'd like to be able to close this issue, we've already had the mailing list to discuss it, haven't we?

#Sorry to mention it to many perople. I'm not sure who is the project owner with CHIRIMEN to have a privilege to mange issues to close/open.

dynamis, 2016-01-25:
We are now using MozOpenHard ML for anything about the project including CHIRIMEN dev:
https://groups.google.com/group/mozopenhard
and Slack for daily discussion:
http://mozoh.slack.com/

Feel free to write in english. We also write in English when we cross post to W3C or mozilla ML etc.
As for auto-translation, sorry for inconvenient but please use google translate or add-ons for it...

closed 2016-01-25 by dynamis

autonome, 2016-02-02:
The mailing list is not open for anyone to join. Is there a reason to not
open it up?

dadaa, 2016-02-02:
Oh, however this google group is "Shared publicly".
https://groups.google.com/group/mozopenhard
Couldn't you read mails?

autonome, 2016-02-02:
ah, yes! i opened in a private browser window and could read them.

i had just subscribed, and the subscription said i had to apply to join. i
hope you will allow me to join. who would not be allowed to join?