MozOpenHard/CHIRIMEN #28: Create #chirimen and #chirimen-dev IRC Room

arky, 2015-08-29:
Please consider creating #chirimen and #chirimen-dev IRC rooms on irc.mozilla.org. The #chirimen is for users and general discussions and #chirimen-dev is for chirimen developer team. 

Thanks!

autonome, 2015-08-29:
Would be great to have just one room for now, until the communication
volume is too overwhelming and we absolutely must have two.

arky, 2015-08-30:
Agree with @autonome Let's keep it simple now.

naokisekiguchi, 2015-08-31:
:+1:  How can we create the room?

naobsd, 2015-08-31:
@naokisekiguchi https://wiki.mozilla.org/IRC#Create_your_own_channel

naokisekiguchi, 2015-08-31:
I have created and registered #chirimen room on irc.mozilla.org. Please try to access.

alexsalas, 2015-08-31:
I'm ready connected with [Instantbird](http://instantbird.com/)

arky, 2015-09-09:
Closing this bug as resolved. Thank you!

closed 2015-09-09 by arky