MozOpenHard/CHIRIMEN #29: Recovery Mode: Provide visual notification using LED blinks

arky, 2015-08-30:
Provide a visual notification using the onboard LED when the Chirimen board enters a recovery mode. At the moment, you require a external display and it is tricky process. 

Usecase: 
1. Hold Reset button and connect the power adapter 
2. As the Chirimen enters the recovery mode, the onboard LED blinks to provide a visual notification

naobsd, 2015-08-30:
well, let's make "recovery mode" clear at first.

----

there is a recovery partition on NAND which contains recovery program. recovery program is used for (e.g.) self(OTA) update, factory reset, etc and generally without PC.
"recovery mode" is the condition that recovery program is running on CHIRIMEN.

recovery program can be automated, so external display (attached to CHIRIMEN) may not be required. but someone must operate CHIRIMEN to enter recovery mode, so external display will be available in most cases.

we can change recovery program behavior easily.

----

and, there is "recover mode" button (no "reset" button) on the board. it can be used to enter "loader mode" for flashing firmware image by flashing tools (AndroidTool.exe, upgrade_tool, etc) on PC.
(there are some other way to enter "loader mode")

PC (with display) is required to operate flashing tools. external display (attached to CHIRIMEN) is not required (nothing is displayed on "loader mode"). flashing tool on PC will tell about CHIRIMEN is in "loader mode" or not.

in this mode, only small bootloader code is running on CHIRIMEN. binary only proprietary bootloader doesn't support LED blinking. only 1 GPIO can be controlled. (well, it's possible to make blinking LED _by external hardware_ controlled by GPIO, but...)

there is somehow working u-boot bootloader for CHIRIMEN too. it's unofficial, no support, and unstable. (we can fix and improve it if someone can work)

----

then, there is GPIO controllable Green LED on CHIRIMEN. currently it's off by default, but

* turn it on by bootloader if possible (-> always on on "loader mode")
* blink it by kernel (-> blinking on "recovery mode")
* turn it off when fxos is started

will be possible and probably easy.

(Red LED is uncontrollable, only for power indicator)

flaki, 2015-08-30:
@naobsd:
> then, there is GPIO controllable Green LED on CHIRIMEN. currently it's off by default, but
>
>    turn it on by bootloader if possible (-> always on on "loader mode")
 >   blink it by kernel (-> blinking on "recovery mode")
 >   turn it off when fxos is started

I think this is a nice and straightforward fix. "blinking" led means system is busy with a "boot sequence", once Firefox OS started blinking should have started. I don't know how much time does the kernel take to boot the OS to a state in which the led could be disabled, but a few seconds sould only result in only a few blinks of the led, then the system should fire up and the led would go dormant.

Obviously if someone have tried to start the system in Recovery mode and waited a few seconds, if the led was still blinking he could be sure that the system is in Recovery mode and could proceed with whatever he tried to achieve in the first place (flashing new system etc...).

closed 2016-07-05 by MSakamaki