MozOpenHard/CHIRIMEN #33: USB camera is not working

naobsd, 2015-12-02:
https://github.com/flingone/hardware-rk29/commit/7a1a610b9f441dd54e7840db7e0b10bcf26c0338 this should be reverted.

naobsd, 2015-12-20:
in addition to hardware/rk29/, hardware/libhardware/ and bionic/ from Rockchip SDK are also required.

naobsd, 2015-12-20:
preview is not working. probably camera HAL need to be fixed.

https://github.com/flingone/hardware-rk29/blob/90003af61aca05a3c5d0404536bb66d1ae35c18d/camera/CameraHal.cpp#L3108
total was fewer than minimum.

I tried "total = numBufs+undequeued;" instead of "total = numBufs;". few preview frames were displayed, but it was not updated after that.

it might be better to check how it works on Android.

dadaa, 2016-02-09:
OpenWebBoard can use USB camera,
http://flatbird.github.io/2014/11/21/owb-camera-api/
So, we may be able to refer OpenWebBoard source code.
http://opensource.kddi.com/owb/owbsource.html

YusukeYamamoto, 2016-02-09:
+1
That's great idea.
 hardware > rk29 > Camera があやしい

closed 2016-07-05 by MSakamaki