MozOpenHard/CHIRIMEN #34: Port to v2.5 from v2.1

YusukeYamamoto, 2015-12-03:
## Step 1 ( only v2.5 )
- [ ] Gaia 
- [ ] Gecko
- [ ] manifests
- [ ] b2g-patches

- [] fix bug at flash.sh ( #31, #13)

## Step2 ( v2.5 + GPIO / I2C patch )
- [ ] Gaia 
- [ ] Gecko

YusukeYamamoto, 2015-12-03:
Done gaia, gecko
https://github.com/MozOpenHard/gaia/tree/v2.5
https://github.com/MozOpenHard/gecko-dev/tree/b2g44_v2_5

autonome, 2015-12-03:
@YusukeYamamoto after you have ported the GPIO/I2C patch, can you attach the individual patch to https://bugzilla.mozilla.org/show_bug.cgi?id=1166588? It will give more people the opportunity to see the code and discuss how to get it landed on mozilla-central.

YusukeYamamoto, 2015-12-03:
@autonome , OK got it. 
Actually, I'm already ready to show the GPIO/I2C patch on FxOS v2.5.
ASAP, the patch will be attached on that bug.
I think, it's useful to discuss on Mozilla work week at Orlando.

YusukeYamamoto, 2015-12-04:
@autonome I've attached the patch on that bug.
https://bugzilla.mozilla.org/show_bug.cgi?id=1166588

YusukeYamamoto, 2015-12-07:
I've added v2.5 branches Gaia/Gecko on MozOpenHard repo.
You can download codes with 
> git clone https://github.com/MozOpenHard/B2G.git
> cd B2G
> BRANCH=mozopenhard-v2.5-no_seccomp ./config.sh chirimen

closed 2016-07-08 by MSakamaki