MozOpenHard/CHIRIMEN #53: Prompt user to acces GPIO/I2C devices

dynamis, 2016-01-25:
任意の Web サイトからユーザの確認もなくハードウェアの操作が可能だが、それでは流石にいろいろとセキュリティ上不味いので、(ユーザに理解不能な確認になるかも知れないが) ハードウェアデバイスを操作して良いですか？というような確認ダイアログを表示して許可された場合に使えるようにする。

note: Permission Table の定義:
https://dxr.mozilla.org/mozilla-central/source/dom/apps/PermissionsTable.jsm

API アクセスを Permission Manager で制限して PROMPT_ACTION にしましょうって話

YusukeYamamoto, 2016-01-26:
実装にあたっての情報展開( 私もよくわかってないので調べながら書いてます )

PermissionTable.jsm に追加は必要だけれど、他に必要な箇所あるのかという観点で調査。

TV Manger API を参考に調査 Bug 998872
これでコミットされているのでこれを見ると早い
https://github.com/mozilla/gecko-dev/commit/d12e06f1df4a99f54af28df13d09f55fc0f34924

PermissionTable.jsm以外にも
Navigator.webidlにも必要そう
https://github.com/MozOpenHard/gecko-dev/blob/9b9f5ba09f340f64910d8c803b6bc5b630ffb1f8/dom/webidl/Navigator.webidl#L361-L367
ここに
https://github.com/MozOpenHard/gecko-dev/blob/9b9f5ba09f340f64910d8c803b6bc5b630ffb1f8/dom/webidl/Navigator.webidl#L361-L367
こんな感じで、checkPermissionの追加。Geckoから提供するIFを列挙。

さらに、IF自体のWebIDLに
https://github.com/MozOpenHard/gecko-dev/blob/9b9f5ba09f340f64910d8c803b6bc5b630ffb1f8/dom/webidl/Navigator.webidl#L361-L367
の様にcheckPermissionを追加

を今のGPIO/I2Cに追加すれば大丈夫そう。
※Prefsについては、別Issueで記載

dadaa, 2016-02-09:
WorkerNavigator に mozGpio および mozI2c を実装したが，Worker における CheckAnyPermissions の実装はまだされていない模様．

closed 2016-07-06 by MSakamaki