MozOpenHard/CHIRIMEN #54: change device name shown in WebIDE

dynamis, 2016-01-25:
Current CHRIMEN is recognized as "Firefox OS (CHIRIMEN)".
We'll change this to just "CHIRIMEN"

dadaa, 2016-01-30:
This "Firefox OS($devicename)" is defined in ADB Helper addon.
https://github.com/mozilla/adbhelper/blob/master/scanner.js#L136
How can we resolve..

autonome, 2016-02-02:
Have you submitted a pull-request to that repo?

dadaa, 2016-02-02:
not yet...
I'll try to pull request to display only device name.

dadaa, 2016-02-03:
I sent pull request to AdbHelper.
https://github.com/mozilla/adbhelper/pull/26

dadaa, 2016-02-04:
This pull request was merged to mozilla/adbhelper.
I close this issue.

closed 2016-02-04 by dadaa

dadaa, 2016-02-05:
The addon with that change was released in version 0.8.7.
https://ftp.mozilla.org/pub/labs/fxos-simulator/adb-helper/